arutar/FrankenDriver #1: Jieshuo performance

Nalryn, 2022-12-16:
3d marks drop from 2350 to 2050 utilizing drivers prescribed vs what jieshuo prescribed as 9 month old drivers. any suggestions? 2d benchmark rose from 613 to 670. same jieshuo 3070m.

Nalryn, 2022-12-16:
less that 1 fps playing 7 days to die. hook me up with how to fix this with your drivers. cause the chinese drivers actually played full 60fps capped. and you stated yours were better.

arutar, 2022-12-17:
I tested the performance of Chinese and original drivers at my stand. It seems that performance jumps from one driver version to another. This is due to the fact that the nvidia developers are constantly improving the driver. In the Frankendriver project, there is no modification of the original driver from Nvidia and you get the performance that was laid down by the Nvidia developers. Added driver version 512.95 which, according to my tests, turned out to be a little more productive than the Chinese one.
[image]
[image]
[image]

closed 2022-12-17 by arutar