arutar/FrankenDriver #2: Easy Anti-Cheat error

arutar, 2022-12-19:
Not working Easy Anti-Cheat crashes with an error.
[image]

arutar, 2022-12-19:
I wrote to EAC support, we'll see what they say. The FrankenDriver project is not a hacker tool, a debugger, or a cheat program. Modifications of the original nvidia binaries do not occur, it is not clear why the EAC does not work.

dynamo-foundation, 2022-12-19:
If you launch games from the NVidia GEForce experience they will run ok.  I was able to start Valorant which uses easy anti-cheat.

ssateneth, 2022-12-19:
Anticheats probably check the signing authority against known good signature sources like verisign or Microsoft Windows Hardware Compatibility Publisher. Since it's self-signed it may not pass some checks, but dynamo and 1 of person is discord have confirmed that some problem games can bypass the error as long as you use geforce experience.

arutar, 2022-12-19:
But all driver binaries are signed by nvidia. And nvldumdx.dll is also signed by nvidia!

Whippy89, 2022-12-20:
Loving all the work you're doing with these drivers, I have a 3070M from
AliExpress. Your driver worked great with most games, but I had an issue
with Back 4 Blood not recognising the graphics card at all - so I had to
reinstall Jieshuo's drivers. Back 4 Blood uses Easy Anti Cheat but I
managed to bypass the nvldumdx.dll error, but when I got into the game it
didn't recognise the GPU and the game ran at around 2fps :( Do you own Back
4 Blood so you can test the game for yourself and maybe find a solution?

dynamo-foundation, 2022-12-20:
> Loving all the work you're doing with these drivers, I have a 3070M from AliExpress. Your driver worked great with most games, but I had an issue with Back 4 Blood not recognising the graphics card [...]

Try running it from GeForce Experience.  Games which I could not otherwise run were ok when launched from that tool as opposed to running from their desktop icons.  GF Exp seems to "wrap" the game somehow so it's trusted.

Whippy89, 2022-12-20:
Didn't work, unfortunately. Back 4 Blood loads up fine with Geforce
Experience with no errors, but when in-game it doesn't recognise the GPU
and runs around 1-2fps... going into the game's settings the Display
Adaptor just shows up blank where it would normally say which GPU you are
using.

Whippy89, 2022-12-20:
And it also doesn't let you select the Display Adaptor from the menu, the
option is just greyed out.

arutar, 2022-12-20:
I was unable to launch Apex Legends. The game does not start via Geforce experience...

dynamo-foundation, 2022-12-21:
Apex Launched without issue.  I was able to go through the entire training mission and entered a match, no errors.

[image]

https://user-images.githubusercontent.com/81651363/208810076-e54d8255-d01e-471d-8791-caabcfd02488.MOV

dynamo-foundation, 2022-12-21:
> Didn't work, unfortunately. Back 4 Blood loads up fine with Geforce Experience with no errors, but when in-game it doesn't recognise the GPU and runs around 1-2fps... going into the game's settings the Display Adaptor just shows up blank where it would normally say which GPU you are

Ok so this game is like $60  - I dont mind testing it - but is it possible to share via steam?  If not I'll go ahead and purchase.  Thanks!

tiagolauer, 2022-12-26:
New World isn't working too! Try to load that and this give me the erro bellow:
[image]

So when I click in "OK", show me that:
[image]

arutar, 2022-12-26:
Now there is Easy Anti-Cheat support!!!

https://github.com/arutar/FrankenDriver/issues/6

closed 2022-12-26 by arutar