arutar/FrankenDriver #3: Will you keep drivers up to date?

ssateneth, 2022-12-19:
Will you keep drivers up to date? There's a guy in USA that has like 400 of the RTX 3060M aliexpress GPU's and they're having a hard time selling them due to driver issues.

Also, are you able to show the "source", or rather, the unpacked version of the latest driver in github? Because some people might be scared of downloading a "shady" driver and worried about viruses or built in crypto miners. I know this driver doesn't have that because you just altered the .inf file to add device ID's, self-signed the entire driver package with your own root certificate, and included the root certificate so that the new signature is trusted by windows. Some setup scripts were altered to include installing the root cert automatically and point to the modified inf file

Lastly, I've seen BIOS's for device ID 2560 for RTX 3060M on techpowerup; will you include that too? I haven't seen any physical devices yet that use 2560 but they supposedly exist.

arutar, 2022-12-19:
Yes I plan to constantly update the drivers.

The unpacked version of the driver takes up a lot of space in the form of binary files, and it is impossible to store binaries in such quantity on Github. Also, if someone wants to see what's inside the archive, it can be done easily with WinRAR and 7zip.

I'm thinking of adding a list of supported devices so that you can add the ID of a new device if it is not in the list.

Can you share a link to information on RTX 3060 with ID 2560?

ssateneth, 2022-12-19:
There are a few BIOS found at https://www.techpowerup.com/vgabios/?architecture=Uploads&manufacturer=&model=&interface=&memType=&memSize=&did=10de-2560-10de-0000&since= but I have not yet seen a physical device.

closed 2022-12-20 by arutar