arutar/FrankenDriver #5: New Device IDs

arutar, 2022-12-20:
This page is for adding new devices that are not in the list of supported.

Sonicwanjie, 2024-06-18:
10DE 2730-1028 0C41, NVIDIA RTX 5000 Mobile Ada Generation.
Hope add this Device ID.

arutar, 2024-06-19:
@Sonicwanjie This device already exists in the regular driver, there is no need to add this identifier.
`NVIDIA_DEV.2730.0C41.1028 = "NVIDIA RTX 5000 Ada Generation Laptop GPU"`

closed 2024-06-19 by arutar

Sonicwanjie, 2024-06-25:
NVIDIA_DEV.2730.0000.10DE，subid is ‘‘0000.10DE’’, how to modify identifier?

arutar, 2024-06-25:
I can add this device, devices, please send me all the device IDs you have and their names.
 example NVIDIA_DEV.2730.0000.10DE      -  RTX 5000 Ada

Sonicwanjie, 2024-06-26:
NVIDIA_DEV.2730.0000.10DE  - NVIDIA RTX 5000 Ada Generation Laptop GPU
NVIDIA_DEV.27BA.0000.10DE  - NVIDIA RTX 4000 Ada Generation Laptop GPU
NVIDIA_DEV.27BB.0000.10DE  - NVIDIA RTX 4000 Ada Generation Laptop GPU

Sonicwanjie, 2024-07-23:
@arutar I miss a vend id，NVIDIA_DEV.28BB.0000.10DE - NVIDIA RTX 2000 Ada Generation Laptop GPU, please add it. Thanks.

Sonicwanjie, 2024-08-27:
NVIDIA_DEV.2717.0000.10DE - NVIDIA GeForce RTX 4090 Mobile

arutar, 2024-11-05:
@Granatoboy3 yes cuda will work fine

sacuro6-crypto, 2025-08-06:
Приветствую, vr (oculus quest 3s )будет работать на 4080m laptop? помнится нужно было раньше 3d vision устанавливать из пакета драйверов, в этих дровах данную строку в режиме установки кастом не видел или оно по умолчанию ставится?

arutar, 2025-08-06:
Несколько людей писали что все ок. Ну по поводу VPN думаю уже прочитали

sacuro6-crypto, 2025-08-06:
arutar. Да про vpn  в курсе , вроде со всеми аспектами ознакомился, вот только вопрос вставал по поводу карт мутантов.

RockDard, 2025-09-15:
NVIDIA_DEV.24B7.0000.10DE - NVIDIA RTX 4000 Ada Generation Laptop GPU

RockDard, 2025-09-15:
[RTX A4000m bios.zip](https://github.com/user-attachments/files/22329298/RTX.A4000m.bios.zip)

NVIDIA_DEV.24B7.0000.10DE - NVIDIA RTX 4000 Ada Generation Laptop GPU

RockDard, 2025-09-15:
PCI\VEN_10DE&DEV_24B7&SUBSYS_000010DE&REV_A1
PCI\VEN_10DE&DEV_24B7&SUBSYS_000010DE
PCI\VEN_10DE&DEV_24B7&CC_030000
PCI\VEN_10DE&DEV_24B7&CC_0300

RockDard, 2025-09-29:
> [@RockDard](https://github.com/RockDard) https://disk.yandex.ru/d/vBRKzy7JbcDVZA

I couldn't find the version with the 'm' designation instead of Frankenstein on the website. Where can I download/find it?

arutar, 2025-09-29:
@RockDard This version of the driver is available only by subscription

mohaxmant-lgtm, 2026-03-29:
You please add new device id
Device RTX A3000 12 GB Laptop GPU

[image]

[image]

chilltrout, 2026-05-14:
@arutar hmm unfortunaley it dosent work i think after 576 the loading of the GSP package changed and this chip wont accept it :(

Sonicwanjie, 2026-08-18:
@arutar 
Pls add DEV ID.
NVIDIA_DEV.2C38.0000.10DE - NVIDIA RTX PRO 5000 Blackwell Generation Laptop GPU

arutar, 2026-08-19:
@Sonicwanjie [New Driver](https://disk.yandex.ru/d/9GPF4_ZtscsAjQ)

diman181074-lgtm, 2026-09-20:
Pls add new device id
Rtx 3050ti mobile
10DE 2523-10DE 0000[image]

arutar, 2026-09-21:
@diman181074-lgtm [New driver](https://disk.yandex.ru/d/gBOlOVc60o_lMg)

evolutiondevel, 2026-09-29:
Add new device please.
On linux works fine on open driver from nvidia repo, but in windows im trying many drivers and  have 43 error :(

[image]

Thanks