arutar/FrankenDriver #7: Displayport

xxxiaomixxx, 2022-12-27:
Displayport does not work on paladin's 3070 laptop after booting into a system with drivers installed. It works in the bios and without drivers

arutar, 2022-12-27:
Does "paladins 3070m" have a different device id?

xxxiaomixxx, 2022-12-27:
Peladn is the vendor of my card, I attached a photo, in the past the T9 message changed it to paladin. I did not try to install the Chinese driver, because games with anti-cheat do not work because of it
[image]
[image]

arutar, 2022-12-27:
Will the image appear if you reconnect the display port?

xxxiaomixxx, 2022-12-27:
No

arutar, 2022-12-27:
Is there any driver with a working display port?

xxxiaomixxx, 2022-12-27:
I had a 512.15 driver (found on YouTube), I bought a new monitor with displayport, I connected it, turned on the computer, the picture was up to the moment of loading into the system, after which it disappeared and the monitor went to sleep. I tried to install your latest driver (527.56), it also installed without errors, but displayport did not work

arutar, 2022-12-27:
Can you send me a screenshot of the gupz?

xxxiaomixxx, 2022-12-27:
[image]

arutar, 2022-12-27:
Have you tried updating the bios on the motherboard and experimenting with booting UEFI/CSM?

xxxiaomixxx, 2022-12-27:
I have a Chinese motherboard (Atermiter X79) for Xeon, there is no more recent bios version on it. It supports UEFI, Windows is installed on the GPT partition. I didn't do anything with CSM

aledsav1, 2023-01-02:
What about flashing the cards bios with another chinese bios, I flashed one of my 3070m with one from techpower, as I had try'd to flash an asus 3070m bios to see if I could work around the driver issue (before arutar modded the driver),  I then just ended up with a slightly bricked card so CH34 came out and I flashed one of the many bioses for these cards back it worked perfectly again.

arutar, 2023-01-13:
@xxxiaomixxx You are not the only one with this problem. It seems that the matter is in the firmware of the video card peladn. I have got a firmware from another Chinese card on which all ports work. Replacing the firmware of a video card is a very risky business. But if the video card does not turn into a brick, then all ports on it will work. If you going to risk, be sure to save the original firmware! I attach a video instruction for updating the firmware and a firmware itself. ALL ACTIONS YOU DO AT YOUR OWN RISK.  
[How to update firmware](https://www.youtube.com/watch?v=4xsEvdIrNGc)
[10DE_0000_FIRMWARE.zip](https://github.com/arutar/FrankenDriver/files/10410783/10DE_0000_FIRMWARE.zip)

aledsav1, 2023-01-14:
Just to add I have flashed one of these card unsuccessfully with an asus laptop bios which was as close to these cards bioses as possible but it did brick the card I think due to the ports (obviously the asus bios does not have 2 hdmi 2 dp) so i then flashed with one of the chinese 3070m bioses on techpower (so not the cards original bios) and it worked no issues

dynamo-foundation, 2023-02-02:
> @xxxiaomixxx You are not the only one with [this problem](https://github.com/arutar/FrankenDriver/issues/12). It seems that the matter is in the firmware of the video card peladn. I have got a firmware from another Chinese card on which all ports work. Replacing the firmware of a video card is a very risky business. But if the video card does not turn into a brick, then all ports on it should most likely work. If you going to risk, be sure to save the original firmware! I attach a video instruction for updating the firmware and a firmware itself.
> 
> ## ALL ACTIONS YOU DO AT YOUR OWN RISK.
> [How to update firmware](https://www.youtube.com/watch?v=4xsEvdIrNGc) [10DE_0000_FIRMWARE.zip](https://github.com/arutar/FrankenDriver/files/10410783/10DE_0000_FIRMWARE.zip)
> 
> Please write here if the flashing of the firmware was successful and the ports are working, or if the video card is broken after flashing.

Do you think this bios will work with 3060m with 1xHDMI and 3xDP?

arutar, 2023-02-02:
This firmware is for 3070m.