arutar/FrankenDriver #8: error 31 in nvidia platforms controller and framework

igoralexeev, 2023-01-04:
Восклицательный знак в диспетчере устройств. но вроде бы игрушки идут, адоб премьер кажется работает

Это устройство работает неправильно, т.к. Windows не удается загрузить для него нужные драйверы. (Код 31)

{Операция не выполнена}
Запрошенная операция не выполнена.

Возникла проблема с запуском устройства ROOT\UNNAMED_DEVICE\0001.

Имя драйвера: oem21.inf
GUID класса: {62f9c741-b25a-46ce-b54c-9bccce08b6f2}
Служба: nvpcf
Нижние фильтры: 
Верхние фильтры: 
Проблема: 0x0
Состояние проблемы: 0xC00000E5

windows  11 22H2  22622.607

arutar, 2023-01-11:
Можно просто удалить это устройство!

closed 2023-01-11 by arutar

arutar, 2023-02-20:
@igoralexeev Сегодня выложу драйвера без этой ошибки

arutar, 2023-02-20:
@igoralexeev Выложил пока 512.95 можно Clean Install сделать и ошибка пропадёт!