arutar/FrankenDriver #12: Black Screen while installing it

marcoscostaf, 2023-01-12:
I'm trying to install this 3070 laptop on windows 11 64bit, when it's installing, it gives a black screen and it doesn't install.  PC reboots to black screen.  Does anyone know how to help?

arutar, 2023-01-12:
Try changing the port on the video card to connect to the monitor (to another hdmi or DisplayPort)

arutar, 2023-01-12:
And with Chinese drivers, a black screen too?

arutar, 2023-01-12:
Do you have a Peladn 3070m?

aledsav1, 2023-01-12:
It is either the card the cards bios or your cables/inputs, I have 2 x 3070m and the display works with this driver no issues and also the chinese drivers, also even if I try a normal nvidia (non modded) driver I still have screen though error 43 so it is not the drivers causing this in anyway

marcoscostaf, 2023-01-12:
- I tried to change the ports.
- the chinese drivers didn't install saying that “ Code execution cannot continue because clb.dll is not found.  Reinstalling the program to fix the problem”
- it is a Peladn 3070m

arutar, 2023-01-12:
It's a video card firmware issue. Without the installed driver, all ports work, but as soon as you install the driver, only HDMI works.

arutar, 2023-01-12:
1. Install driver
2. Connect monitor via HDMI  
   (If the monitor does not have an HDMI connector, then you need an adapter (HDMI - monitor input port))

arutar, 2023-01-12:
You can also try to fix the firmware of the video card.  
Can you post the firmware from your video card here?

arutar, 2023-01-12:
You can run gpu-z and save the firmware, and post it here

marcoscostaf, 2023-01-12:
[GA104aliexpress.zip](https://github.com/arutar/FrankenDriver/files/10404809/GA104aliexpress.zip)

arutar, 2023-01-12:
@aledsav1 Can you upload the firmware from your RTX 3070m (do you have 4 ports on your video card? 2 hdmi 2 DIsplayPort?)

aledsav1, 2023-01-13:
Yes I do have 2 x 2, I will upload shortly

aledsav1, 2023-01-13:
this should be the bios for my 3070m

aledsav1, 2023-01-13:
couldn't upload bios rom here, tried sending via email, but that is also just to here? maybe you have

arutar, 2023-01-13:
You can share a link to Google drive

aledsav1, 2023-01-13:
https://drive.google.com/file/d/1qUZ200TmUphDI5Rf-7wzHhWO0NLO2Nd5/view?usp=sharing

arutar, 2023-01-13:
@marcoscostaf  
I wrote [here](https://github.com/arutar/FrankenDriver/issues/7) a risky instruction for flashing the firmware

closed 2023-01-25 by arutar