arutar/FrankenDriver #14: ReBAR 3060m

StalkingMyPerfil, 2023-01-14:
Hi I'd like to ask if Resizable bar is possible on RTX 3060M? I've been looking around for a way to get it enabled on my BIOS but it just doesn't work
do U know if is possible? if u do so could u tell me how to enable it Thanks !

wingcobob, 2023-01-17:
Last I looked into it the vbios has to support rebar for it to work, if you look in nvidia control panel -> system information it will tell you if the card supports it. I know my 3070m doesn't

arutar, 2023-01-17:
Gpu z has determined that there is rebar support!   
I will test the possibility of enabling it in the near future.
[image]
[image]

arutar, 2023-01-21:
The resizable bar is not supported by the firmware. You can try to request the firmware from the manufacturer. And open a dispute about an unsupported feature.

arutar, 2023-01-23:
I unsuccessfully flashed my card several times in search of a suitable firmware with support for resizable bar.

closed 2023-01-25 by arutar

3nly, 2023-09-23:
Is there really no way to get reBAR support on these cards? Feels like it's more of a driver problem than anything else.

arutar, 2023-09-23:
@3nly The resizable bar is not supported by the firmware.