arutar/FrankenDriver #20: Support for 3060m with 1 HDMI and 3 DP

dynamo-foundation, 2023-02-01:
Greetings, happy to pay you to assist with getting these to work.  Currently the HDMI output is only active; display ports have no output.  Driver works great otherwise.  This card is PCWINMAX brand.  I have hundreds.  Looking to sell some but I want to get the DP outputs functional first.

Thanks!

aledsav1, 2023-02-02:
Hi don't know if you are 'bargainclubshop'? as they are selling the exact same card without dp working. I can only assume your bios is not configured for the DP ports or they have done something hardware related different with this version of the card, I am sure arutar will jump in, but I think your only solution is to try the bioses on techpower for the aliexpress versions of these cards and see if one enables the dp.

arutar, 2023-02-02:
firmware from my 3060m with all working ports. [Firmware](https://github.com/arutar/FrankenDriver/files/10290198/firmware.zip)

dynamo-foundation, 2023-02-02:
Thank you for posting this :)  Unfortunately the driver will not start with this firmware flashed (code 43).

GPU-Z can see the card, but it reports 8GB of RAM - these are 6GB cards.

[image]
[image]
[image]

dynamo-foundation, 2023-02-02:
> Hi don't know if you are 'bargainclubshop'? as they are selling the exact same card without dp working. I can only assume your bios is not configured for the DP ports or they have done something [...]

ya the problem is there are hundreds and this one can only be flashed with 341a, so you have to take apart, flash, reassemble, test - it could take months lol

aledsav1, 2023-02-03:
> ya the problem is there are hundreds and this one can only be flashed with 341a, so you have to take apart, flash, reassemble, test - it could take months lol

I think that you can flash 1st bios attempt with nvflash, then if it bricks use a ch341a, until you get the right bios, then you know oyu can flash the others straight from nvflash...so yes it does mean 1 needs to be dimantled and ch341a used, but only on one.
As you have hundreds I think it is worth the time too do on one card so you can then sell the cards with working ports, though these cards are a hard sell to the non mining market in the volume you have, I only had 2 3070m and they did not sell, I had to make gaming pcs with them in...the guy I bought them from had many and they were listed on ebay for a long time with only a few selling in months, his listing is gone so I think he had someone buy them in bulk?

aledsav1, 2023-02-03:
> Thank you for posting this :) Unfortunately the driver will not start with this firmware flashed (code 43).
> 
> GPU-Z can see the card, but it reports 8GB of RAM - these are 6GB cards.
> 
> [image]
> 
> [image]
> 
> [image]

this is the 3070 bios that is why it is showing 8g! obviously an error with the bios version you need the 3060 aliexpress bioses

arutar, 2023-02-03:
@aledsav1  @dynamo-foundation The firmware that I posted should be version **(94.06.01.27.01)** and you have a different version after the flash. It looks like you flashed the wrong firmware. (94.04.5A.00.90) is the firmware from 3070Ti

dynamo-foundation, 2023-02-03:
> @aledsav1 @dynamo-foundation The firmware that I posted should be version **(94.06.01.27.01)** and you have a different version after the flash. It looks like you flashed the wrong firmware. (94.04.5A.00.90) is the firmware from 3070Ti

Is there a byte location in the firmware that shows the version number?  I'll search later with hex editor.  I'm very confident I flashed the one from the link.  Anyway I can try again this evening.

I do also have a 3070Ti in this system.  It's the only way I can get it to boot because the other card has no display output.

Thanks!

dynamo-foundation, 2023-02-03:
> 

cool I will try nvflash once I get a working version

dynamo-foundation, 2023-02-10:
A few learnings, but no real progress.

In order to burn this NOR flash you have to erase it first.  At least using ASProgrammer.

I was finally able to get the image installed and verified the VBIOS signature.  The version number appears in numerous places in little Endian format (e.g. you can search for 01 27 01 06 94 in the hex code and see it).  So I know for sure the firmware you posted above is indeed loaded to the card.

Using this card (along side my 3070ti), I can get my desktop to boot about 50% of the time.  The other 50% it hangs, then bluescreen and goes into recovery mode and says I have to rebuild the OS.  Removing the 3060 card and rebooting seems to fix it.

When I can get the desktop to boot, either the card wont start (Code 43) or the card is not found at all, and instead goes into "unknown PCI device" slot in device manager.  Running TPU reports the wrong VBIOS version (the 3070) even though I have double checked the hex loaded to the VBIOS does not have that version, so I suspect there is just an error in reading the version.

I'm going to try one other VBIOS that I found here:
https://www.techpowerup.com/vgabios/246486/246486
This seems to be the "other" variant of this card from Jieshuo.

Also note this VBIOS does not support UEFI so you need to set CSM in order for it to POST on this card at all, which means you need a secondary card to bootstrap into the motherboard BIOS to set CSM in the first place.

dynamo-foundation, 2023-02-10:
Several of these BIOS images will boot the card, however none of them enable the Display Port interfaces.  I'm giving up on this for now - if anyone else sorts this please ping me.  Thanks!

closed 2023-02-10 by dynamo-foundation