arutar/FrankenDriver #21: FPS IN-GAMES Like Minecraft, Roblox

StalkingMyPerfil, 2023-02-08:
is it normal to get a LOW FPS amount? I mean it doesn't change when I'm at max settings and low settings
I used to get 258 FPS on low settings and like 70 FPS on max settings on a GTX 1050 TI 4 GB
when I got the RTX 3060M I'm getting 160 on max Settings and 170 on low settings

is there any adjustments I can do or smth else?

arutar, 2023-02-08:
Do you have vertical synchronization enabled? try disabling it.  
Is ray tracing enabled?

StalkingMyPerfil, 2023-02-08:
> 

Nope They don't have any of these options enabled

StalkingMyPerfil, 2023-02-09:
it helped a bit but I don't know if this is because Roblox isn't optimized for the GPU because when I had a GTX 1050 TI I used to get around 270 FPS on low settings but with this one I'm getting only around 202 but thanks alot for the help!

closed 2023-02-09 by arutar

dynamo-foundation, 2023-02-10:
My experience with the 3060s I have has been quite good in terms of frame rate.  All of the games I tried to run were totally fine and close to the performance of my 3070TI.