arutar/FrankenDriver #22: EAC problems

toastie000, 2023-02-20:
https://streamable.com/beopig
FDriver = FrankenDriver

The streamable clip will explain the issue, but I'll share some more context. The latest FDriver had been working fine for the past 3 weeks, but I was getting awful fps in RDR2 and Tarkov and was curious to know if a previous FDriver would fix this (it did not FYI). I did that yesterday and even played an EAC game after reinstalling the latest FDriver, but today it decided it wanted to have problems.

I didn't test 528.02, but I did test 512.95 and 528.24. I reinstalled 528.24 multiple times, and was uninstalling with DDU of course. I am using a 3070m 10DE.249D 10DE.0000.

arutar, 2023-02-21:
Have you read the [instruction](https://github.com/arutar/FrankenDriver/issues/6)

toastie000, 2023-02-21:
Ah, yes. I completely forgot about that when reinstalling the drivers. My apologies.

toastie000, 2023-02-21:
> Have you read the [instruction](https://github.com/arutar/FrankenDriver/issues/6)?

Ah, yes. I completely forgot about that when reinstalling the drivers. My apologies.

closed 2023-02-21 by toastie000