arutar/FrankenDriver #24: Dynamic Super Resolution

wOOOlfykt2, 2023-03-06:
Hello,

I am writing to inquire about the Dynamic Super Resolution feature for the NVIDIA RTX 3060M with your unofficial driver for personal computers. Can you confirm whether this feature is supported on this video card with your driver?

If you have any information on this topic, please let me know. Additionally, if you have any advice or recommendations for how to best utilize the Dynamic Super Resolution feature with the NVIDIA RTX 3060M, I would be very grateful to hear them.

Thank you for your time and assistance.

arutar, 2023-03-06:
[image]

arutar, 2023-03-07:
@wOOOlfykt2 As you can see from the screenshot, all functions are supported including DLDSR.

wOOOlfykt2, 2023-03-08:
Thank you for your response and for confirming that the Dynamic Super Resolution feature, including DLDSR, is supported on the NVIDIA RTX 3060M with your unofficial driver.

Although I don't have plans to purchase this graphics card at the moment, I appreciate your information and will keep it in mind for any future considerations.

Thanks again for your time and assistance.

closed 2023-03-09 by arutar