arutar/FrankenDriver #25: LiveKernelEvent - 141

toastie000, 2023-04-05:
I'm using a 3070m on 94.04.35.00.25 BIOS and driver 531.26. I've used the other two drivers and had the same problem.

I'm getting this error at least once every time I use my computer. My screen tends to just temporarily go black whilst it restarts the driver automatically, at least that's what I'm assuming. What's causing the driver to restart, I don't know. But, sometimes it will completely go black and force me to restart the computer as I cannot manually restart the driver with Win + CRTL + Shift + B. There's of course more examples in Reliability Monitor, but I don't wish to flood the thread with the same photos.

[image]
[image]
[image]

arutar, 2023-04-05:
@toastie000 It looks like you have a hardware problem. Perhaps even with the video card itself (try changing the thermal paste to MX-4). Try flashing the latest BIOS on your motherboard, and run memory tests, check for hard drive errors. Usually no one who uses this driver has problems like you have.

toastie000, 2023-04-05:
I've had FurMark running for the past hour, so it's probably not the memory. Everything ran fine when I had my 1080 Ti and 970 in the system, so it won't be due to a bad Windows installation or drive problems. When I got the card, I installed Kryonaut (thermally better than MX-4, slightly). But I'll update the BIOS and maybe dual-boot Manjaro and see if the issue persists. If it does whilst running Manjaro, then it's probably due to the card itself. I'll leave the thread open for the time being and update in a day or two, unless you have any questions or advice.

toastie000, 2023-04-09:
Forgot to reply, my apologies. After updating BIOS, I also switched which DisplayPort I was plugged into and I don't seem to be getting any errors any longer. Whether it's permanent or if the BIOS update/switching ports 'fixed' the problem, I'm not sure. Anyhow, I shall close this thread.

closed 2023-04-09 by toastie000