arutar/FrankenDriver #28: 3060m

feixiong-cc, 2023-05-09:


feixiong-cc, 2023-07-28:
> 你能提供完整的设备 ID 吗？gpu-z f（或示例）的屏幕截图

[image]

feixiong-cc, 2023-07-28:
I tried all the drivers on the market and it didn't work, it didn't work in time to install successfully, except for this 458.48, can't use the new driver, can you help me?

arutar, 2023-07-28:
@feixiong-cc Try this version of [DRIVER](https://drive.google.com/file/d/17QDhHeRtwWoha_C1H8joPmGPuizNlVi_/view?usp=sharing) And after you check if the driver works, please write here

feixiong-cc, 2023-07-29:
[image]
This driver is still unusable, it can be installed normally without a signature. But it cannot be called.

feixiong-cc, 2023-07-29:
@arutar

arutar, 2023-07-29:
@feixiong-cc As I can see the driver has been installed normally. Do you have multiple video cards in your system? What exactly is not working?

arutar, 2023-07-29:
@feixiong-cc Can you send a device manager screenshot?

feixiong-cc, 2023-07-29:
@arutar The video card only has NVIDIA Graphics Device, yes, it seems to be installed, but it does not work.     I have tried to install more than 20 versions, but it still does not work, except for the version 27.21.14.5848, it can be normal only when it is in system test mode or when the driver signature is disabled.     Any other version I have installed can be installed normally, but it cannot be used normally,

feixiong-cc, 2023-07-29:
@arutar 249f this driver ID, because he doesn't have the right signature I guess. So it can't be used, that's my guess. 458.48 contains a signed certificate, so it can be used normally.

feixiong-cc, 2023-07-29:
[image]

feixiong-cc, 2023-07-29:
[image]

arutar, 2023-07-31:
I can't figure out what exactly isn't working. In order to check everything and correct the driver, I need such a card.

arutar, 2023-07-31:
This device is not a RTX 3060!!!

[image]

arutar, 2023-10-05:
If you have any additional information please let me know. A2000m version has been added for now

closed 2023-10-05 by arutar

feixiong-cc, 2023-10-06:
Thank you very much for your help, I have found out the problem, this graphics card is the inner version of the GA104 for RTX A3000M, it is not available on the market, I think you just need to add my device ID to the updated version, it will work correctly, my device ID is, 10DE.249F.3A4DD17AA

arutar, 2023-10-06:
@feixiong-cc I added support for your A3000m version. Try this version of the [DRIVER](https://drive.google.com/file/d/1oAsWkl-qD2pH8_DaF9w7sYuDLsxijY1F/view?usp=sharing). And after you check if the driver works, please write here

feixiong-cc, 2023-10-07:
Unfortunately, it is still the same, it can install the driver, but it cannot be used normally, and only version 458.48 can be used normally. Thanks for your help

arutar, 2023-10-12:
Next driver version, please try it should work [https://drive.google.com/file/d/1U9VxLcJqxjQ8ZUMMEk5ubUncOWVxP518/view?usp=sharing](DRIVER)

feixiong-cc, 2023-10-16:
It still doesn't work properly, I switched back to the 458 driver it is normal, you can download this [driver](https://drive.google.com/file/d/1NwlPudAYqd_UUu6ATMvJ8liy0o2B_SFe/view) to see if you can find the problem. Thank you so much!

play-day-and-night, 2024-05-25:
> 还是不能正常工作，我切换回458驱动是正常的，你可以下载这个[驱动](https://drive.google.com/file/d/1NwlPudAYqd_UUu6ATMvJ8liy0o2B_SFe/view)，看看能不能找到问题。非常感谢！

can you give me the 249f vbios, send to 1532288474@qq.com,thank you