arutar/FrankenDriver #31: Two 3060m in system

refflex, 2023-09-06:
Hi
In Windows drivers 537.13 are successfully installed. But after reboot black screen. The same situation is with the drivers from the manufacturer. Any chance to fix this?

arutar, 2023-09-06:
@refflex Try connecting your monitor with a different HDMI slot.

arutar, 2023-09-06:
@refflex Will it be the same if you leave only the second card?

refflex, 2023-09-06:
@arutar The cards are flashed with the same BIOS from the manufacturer. Single cards work without problems. I have not tried flashing cards with other bios.

arutar, 2023-09-06:
@refflex It turns out that the cards work without problems if they are connected one at a time, but if you connect two at the same time, then one of them stops working?

refflex, 2023-09-06:
@arutar Yes. More precisely, both cards work until the system is rebooted after installing the drivers. And the second card does not support OpenCL and CUDA (this is important).

arutar, 2023-10-05:
Everything works perfectly on my system.
[image]
[image]
[image]

closed 2023-10-05 by arutar