arutar/FrankenDriver #32: Procedure to build drivers

dynamo-foundation, 2023-10-08:
Greetings!  Can you post a link to the procedure that you use to create these drivers?  I think there is some confusion regarding how to self-sign a driver with the included device ID.  Thanks!

arutar, 2023-10-08:
@dynamo-foundation To create this driver, I use a lot of third-party utilities. Utilities for signing the driver, special modules for patching, separate utilities for archiving and assembly. The driver creation procedure is a command to which the archived driver from nvidia is passed, as a result, a ready-made driver is obtained. The procedure takes about 10-15 minutes, only self-written modules are used for patching the driver

arutar, 2023-10-14:
Performance has been increased since version 537.58

closed 2023-10-14 by arutar