arutar/FrankenDriver #33: Drivers on Mining gpu?

Skylord4321, 2023-10-13:


arutar, 2023-10-13:
So far, driver is not support for (P106-090 / P106-100 / P104-100 / P104-101 / P102-100 / CMP 30HX / CMP 40HX / CMP 50HX / CMP 70HX / CMP 90HX ). But perhaps in the near future I will add this support

Skylord4321, 2023-10-13:
great thanks for telling me, I want to help but unfortunately, I lack experience with modding drivers, however, there is another person on github who has been creating modded drivers for the mining cards (they are working well but still lacking features). So I hope this provides you with some pieces of information about it.
https://github.com/dartraiden/NVIDIA-patcher

arutar, 2023-10-14:
This is a good project and there is even anti-cheat support. So far, I don't see the point in adding these cards to the frankendriver

closed 2023-10-14 by arutar