arutar/FrankenDriver #36: 3060m in hiveos throws same errors as regular drivers

CYRiXplaysHalo, 2023-12-06:
Trying to get a 3060m 6gb to properly load nvidia drivers and tried both free/public versions of the linux drivers in hiveos and the results aren't any better. I have tried the card in multiple slots. I have tried a regular 3070 and it works fine. 

[image]

Log:

`Dec 06 03:20:10 rig0F548A kernel: nvidia: loading out-of-tree module taints kernel.
Dec 06 03:20:10 rig0F548A kernel: nvidia: module license 'NVIDIA' taints kernel.
Dec 06 03:20:10 rig0F548A kernel: Disabling lock debugging due to kernel taint
Dec 06 03:20:10 rig0F548A kernel: nvidia: module verification failed: signature and/or required key missing - tainting kernel
Dec 06 03:20:10 rig0F548A kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 243
Dec 06 03:20:10 rig0F548A kernel: 
Dec 06 03:20:10 rig0F548A kernel: nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
Dec 06 03:20:11 rig0F548A kernel: NVRM: loading NVIDIA UNIX x86_64 Kernel Module  525.85.05  Sat Jan 14 00:49:50 UTC 2023
Dec 06 03:20:11 rig0F548A kernel: nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  525.85.05  Sat Jan 14 00:40:03 UTC 2023
Dec 06 03:20:11 rig0F548A kernel: [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
Dec 06 03:20:11 rig0F548A kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1423)
Dec 06 03:20:11 rig0F548A kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
Dec 06 03:20:11 rig0F548A kernel: [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to allocate NvKmsKapiDevice
Dec 06 03:20:11 rig0F548A kernel: [drm:nv_drm_probe_devices [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to register device
Dec 06 03:20:11 rig0F548A nvidia-persistenced[902]: Started (902)
Dec 06 03:20:11 rig0F548A kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1423)
Dec 06 03:20:11 rig0F548A kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
Dec 06 03:20:12 rig0F548A nvidia-persistenced[902]: device 0000:01:00.0 - failed to open.
Dec 06 03:20:12 rig0F548A kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1423)
Dec 06 03:20:12 rig0F548A kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
Dec 06 03:20:12 rig0F548A systemd[1]: Started Hive Watchdog Service.
Dec 06 03:20:12 rig0F548A hive-watchdog[916]: Watchdog disabled in config (/hive-config/rig.conf)
Dec 06 03:20:12 rig0F548A hive-watchdog[916]: MAX LOAD AVERAGE:     22.0 (min 4.0, def 22.0)
Dec 06 03:20:12 rig0F548A hive-watchdog[916]: OK LA(5m): 0.72 < 22.0, LA(1m): 2.56 < 44.0
Dec 06 03:20:12 rig0F548A kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1423)
Dec 06 03:20:12 rig0F548A kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
Dec 06 03:20:12 rig0F548A kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x24:0x72:1423)
Dec 06 03:20:12 rig0F548A kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
Dec 06 03:20:12 rig0F548A hive[779]: nvtool error (12)`

arutar, 2023-12-06:
Try running it on a test bench separately in Ubuntu. Is the card itself working? Usually everything works well.

arutar, 2023-12-06:
Found this
https://docs.nvidia.com/networking/display/mlnxofedv23070500/uefi+secure+boot

arutar, 2023-12-06:
@CYRiXplaysHalo I found what the problem is. Secure Boot is enabled on the computer where the cards are installed. You need to disable secure boot in the bios and select the unsigned driver installation mode during installation.

There is also a more difficult option, to figure out how to sign the driver in accordance with [this manual](https://download.nvidia.com/XFree86/Linux-x86_64/450.80.02/README/installdriver.html).

CYRiXplaysHalo, 2023-12-06:
@arutar what do I do to select unsigned driver installation mode? That's not on by default with these frankendrivers?

arutar, 2023-12-07:
@CYRiXplaysHalo just reinstall the driver

CYRiXplaysHalo, 2023-12-08:
nothing changes with a reinstall. this is the command I use

```nvidia-driver-update "https://drive.google.com/uc?export=download&confirm=no_antivirus&acknowledgeAbuse=true&id=1ErN4_upq1C5KkJs2EnA9PMQGKQk62Csl"```

Is there some other way to do this?

arutar, 2023-12-08:
```
wget --output-document=NVIDIA-Linux-x86_64-535.104.05.run 'https://drive.google.com/uc?export=download&confirm=no_antivirus&acknowledgeAbuse=true&id=1ErN4_upq1C5KkJs2EnA9PMQGKQk62Csl'
sudo chmod a+x ./NVIDIA-Linux-x86_64-535.104.05.run
./NVIDIA-Linux-x86_64-535.104.05.run
```

CYRiXplaysHalo, 2023-12-09:
yeah that didn't work either.

```
root@rig0F548A:~# wget --output-document=NVIDIA-Linux-x86_64-525.85.05.run  'https://drive.google.com/uc?export=download&confirm=no_antivirus&acknowledgeAbuse=true&id=1Uk8Mw2TcGgqBxYsmBa8EOiKlTSSmn9SS'
--2023-12-09 04:44:34--  https://drive.google.com/uc?export=download&confirm=no_antivirus&acknowledgeAbuse=true&id=1Uk8Mw2TcGgqBxYsmBa8EOiKlTSSmn9SS
Resolving drive.google.com (drive.google.com)... 142.251.32.14, 2607:f8b0:4009:81a::200e
Connecting to drive.google.com (drive.google.com)|142.251.32.14|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://doc-0k-2k-docs.googleusercontent.com/docs/securesc/ha0ro937gcuc7l7deffksulhg5h7mbp1/v7qdpdc0djm2i66knbbv616vd3r29plo/1702089825000/18266593544375721451/*/1Uk8Mw2TcGgqBxYsmBa8EOiKlTSSmn9SS?e=download&uuid=2efdc72e-f9c3-4dbc-be89-f27e7e8f3daa [following]
Warning: wildcards not supported in HTTP.
--2023-12-09 04:44:34--  https://doc-0k-2k-docs.googleusercontent.com/docs/securesc/ha0ro937gcuc7l7deffksulhg5h7mbp1/v7qdpdc0djm2i66knbbv616vd3r29plo/1702089825000/18266593544375721451/*/1Uk8Mw2TcGgqBxYsmBa8EOiKlTSSmn9SS?e=download&uuid=2efdc72e-f9c3-4dbc-be89-f27e7e8f3daa
Resolving doc-0k-2k-docs.googleusercontent.com (doc-0k-2k-docs.googleusercontent.com)... 142.250.191.161, 2607:f8b0:4009:803::2001
Connecting to doc-0k-2k-docs.googleusercontent.com (doc-0k-2k-docs.googleusercontent.com)|142.250.191.161|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 415131327 (396M) [application/x-shellscript]
Saving to: ‘NVIDIA-Linux-x86_64-525.85.05.run’

NVIDIA-Linux-x86_64-525.85.05.run                                   100%[==================================================================================================================================================================>] 395.90M  4.69MB/s    in 84s     

2023-12-09 04:45:59 (4.73 MB/s) - ‘NVIDIA-Linux-x86_64-525.85.05.run’ saved [415131327/415131327]

root@rig0F548A:~# sudo chmod a+rx ./NVIDIA-Linux-x86_64-525.85.05.run 
root@rig0F548A:~# sudo ./NVIDIA-Linux-x86_64-525.85.05.run 
Verifying archive integrity... OK
Uncompressing NVIDIA Accelerated Graphics Driver for Linux-x86_64 525.85.05...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................Extraction failed.

Signal caught, cleaning up

root@rig0F548A:~# sudo ./NVIDIA-Linux-x86_64-535.104.05.run
tail: error writing 'standard output': Broken pipe
Verifying archive integrity... OK
Uncompressing NVIDIA Accelerated Graphics Driver for Linux-x86_64 535.104.05........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................Extraction failed.

Signal caught, cleaning up
```

arutar, 2023-12-09:
@CYRiXplaysHalo you have **Extraction failed** error

CYRiXplaysHalo, 2023-12-12:
yeah this is still an issue...

arutar, 2023-12-12:
@CYRiXplaysHalo Have you checked your hardware for errors?

closed 2024-01-08 by arutar