arutar/FrankenDriver #37: Memory detecting

3Jlou4eJluk, 2023-12-22:
I am using rtx 3070m 16 gb for Machine Learning. There is some trouble with memory. PyTorch can't load models with size above 7.7gb. Is it possible, that this is a driver problem? I used Linux [535.104.05](https://drive.google.com/uc?export=download&confirm=no_antivirus&acknowledgeAbuse=true&id=1ErN4_upq1C5KkJs2EnA9PMQGKQk62Csl).

3Jlou4eJluk, 2023-12-22:
System monitor shows, that 14.4gb of vram in use. I thought that the video card would just be flashed to 16 gigabytes, when in fact there are 8.

arutar, 2023-12-22:
@3Jlou4eJluk I think that everything is fine with the driver and the amount of memory is determined correctly, but there is no rtx 3070 with 16 GB of memory in the desktop version, respectively, you need to edit the source code of the application so that it determines the correct amount of memory.

3Jlou4eJluk, 2023-12-22:
@arutar thak you, for answer!

closed 2023-12-23 by arutar