arutar/FrankenDriver #38: AI LLM models locally on cards like the 3060M with FrankenDriver

WiktorD75, 2024-01-05:
Could someone check if AI models can be run locally on cards like the 3060M 12GB with FrankenDriver? The easiest way is to try using the stand-alone LMstudio.ai application (closed source), but if someone has experience in running LLM locally on open source applications, they can try do it using e.g. oobabooga.
Do you get any errors or do they start normally?

jwadow, 2024-01-25:
Hello again, my friend of interest, I was waiting for a package to answer you.
Today I received the package, and I’ve been playing all evening on Windows 11 Pro with a white _anime_ DICASVER 3070M 16GB, everything works as it should, LLMs work very quickly, 7B models are very fast compared to 1070, perhaps faster than ChatGPT (the quality of the answers does not count).
I tested both mistral instructions v0.1 Q4_K_M and v0.2 Q8_0.
I also tested AUTOMATIC1111 with SDXL, it works. I tried 1.5, 2.1 and a bunch of similar models, everything was very fast.
In Python, training keras tensorflow is also very fast. This is about neural networks. In general, the drivers are great.

Jithub123321, 2024-01-26:
I have a 3070m 16gb and can confirm SD A1111 / ComfyUI work well (on latest drivers), its the reason i got my card specifically. Not had any issues with OOM. Still learning but not ran into any issues with the GPU at all. 70deg on 100% full load 120watts

Jithub123321, 2024-01-29:
> thnx guys, this is great news, but where did you buy your 3070m 16gb?

No worries.

So bit of an strange one. I happen to come across an eBay listing. Could
not find that specific vram arrangement anywhere else. Others all had 8gb.
Too a gamble and it works a treat.

I will link you additional ones. Fyi these won't be the exact same seller I
purchased from though.

https://www.ebay.co.uk/itm/176112977825?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=l8uZ9Eo8Qsy&sssrc=4429486&ssuid=iriokuzpsty&var=&widget_ver=artemis&media=COPY

The ram works with no issues, I have carried out tasks to utilise it fully.

closed 2024-02-25 by arutar