arutar/FrankenDriver #39: 3070M 16 GB VRAM

jwadow, 2024-01-07:
On marketplace OZON I liked the white 3070 with 16 GB of video memory for $290 by "DICASVER" brand. First of all, I was interested in the presence of 16 gigabytes of video memory, because I can run much more, for example SDXL.
I need a working ShadowPlay in Geforce Experience.
Games don't interest me much, but as I see, the problems with anti-cheats have been resolved, so I can play whatever I want whenever I want, for example CS 2? I would like to know about functionality of neural networks like Stable Diffusion, about Python and tensorflow/torch, about Blender, about Davinci Resolve, about Nvidia's AI chromakey for webcam, about RTX Voice. I definitely need CUDA computing.

IN GENERAL, I would like a fully functioning video card on both Windows and Linux.

What common disadvantages do these drivers have, given that I want a fully functioning video card for any possible task, like the original Nvidia?

arutar, 2024-01-08:
@jwadow These drivers have no problems and they are no different from the original drivers. You won't have any problems with ShadowPlay, Geforce Experience, Blender, Davinci Resolve. But most likely there will be difficulties with neural networks since they are optimized for specific video card models and there is no Mobile version of the 3070 with 16gb of memory (most likely you will have to edit the sources of Ai applications to fully use all the memory of the video card).

arutar, 2024-01-08:
@jwadow If you buy this video card, please share your experience. There may also be problems with the card itself (there were several early reviews with an attempt to increase memory to 16gb on the RTX 3070 and these cards did not work correctly under certain conditions)

closed 2024-01-18 by arutar

jwadow, 2024-01-25:
> @jwadow If you buy this video card, please share your experience. There may also be problems with the card itself (there were several early reviews with an attempt to increase memory to 16gb on the RTX 3070 and these cards did not work correctly under certain conditions)

Today I received the package, and I've been testing all evening on Windows 11 Pro with a white anime DICASVER 3070M 16GB, everything works as it should, LLMs work very quickly, 7B models are very fast compared to original turbine 1070 8GB, perhaps faster than ChatGPT (the quality of the answers does not count).
I tested both mistral instructions v0.1 Q4_K_M and v0.2 Q8_0.
I also tested AUTOMATIC1111 with SDXL, it works. I tried 1.5, 2.1 and a bunch of similar models, everything was very fast, and Batch Size directly affects video memory, that is, it is not fake and works.
In Python, training keras tensorflow is also very fast, and all the necessary video memory is working.

Also checked ShadowPlay, Nvidia Broadcast, NVENC and everything else. Everything is working.

I also tested games with RTX like Quake 2, everything works great. I tested the Furmark hairy donut, no problems. Even the leaked STALKER 2 worked, and all my 16 gigabytes worked there. Of course, I haven't dived into all sorts of games like RDR2 or others, but all my necessary tasks work perfectly. 16 gigs work everywhere I need them (although 3070 by 16 does not exist).

Even if somewhere 16 gigabytes do not work, I will treat it as if the application is running on 1 processor thread, although I have 16 of them.

Thanks for such a wonderful tool. Although I won’t donate, for this reason I bought this video card, I will recommend Frankendriver to everyone passing by. I hope at least my messages will make a small contribution to uncertain, but future users of Frankenstein.