arutar/FrankenDriver #40: China 3070M + Original 1070 - work together?

jwadow, 2024-01-08:
At the moment I have an original reference turbine 1070 8 GB. As I wrote in another issue, on OZON marketplace I liked the white 3070 with 16 GB of video memory for $290 by "DICASVER" brand.

Most likely I will buy it and share my impressions, but I would like to install 2 video cards at the same time, that is, the original 1070 8GB and the chinese 3070M 16GB.
We're not talking about SLI, of course.

What about workability? Before that I installed 650+760, I also installed 760+1070, and everything worked. But all these video cards were original with original drivers. But what about now?

arutar, 2024-01-08:
Everything will work fine. But you need to install Frankendriver on original 1070 and on the 3070m at the same time

closed 2024-01-08 by arutar

jwadow, 2024-01-08:
> Everything will work fine. But you need to install Frankendriver on original 1070 and on the 3070m at the same time

The fact of the matter is that the driver must stand alone. After all, you cannot install 2 Nvidia installations at the same time.

Therefore, I need to remove all the old drivers, do a clean install of FrankenDriver and it will work on my 1070? And therefore two video cards will work?

arutar, 2024-01-09:
@jwadow When the driver is installed, it is installed on all supported nvidia cards on your computer

jwadow, 2024-01-25:
> @jwadow When the driver is installed, it is installed on all supported nvidia cards on your computer

I couldn't start my 1070+3070M. I've been on FrankenDriver for 2 weeks on my original 1070, just for fun. Today I inserted the 3070M instead, everything worked right away. But as soon as I plugged the 1070 into the second slot, my 3070M was without a driver at low-res like 640x480 until I pulled out my 1070.
Manipulating drivers and reinstalling did not help. So I’ll have to give up my 1070, but I wanted to expand the memory for some of my tasks :(