arutar/FrankenDriver #45: Showkings RTX 3060m 6gb fans run 100% and no output from the displayports?

Usulyre, 2024-02-23:
Hi, i bought a showkings rtx 3060m and the fans run 100% all the time and the displayports don't work.

I have installed your driver but it didn't help those issues.

Any bios update would help this?

arutar, 2024-02-23:
Here is the original bios on jieshuo 3060. You make all modifications at your own risk.
[JS_ORIGINAL_GA106.zip](https://github.com/arutar/FrankenDriver/files/14389750/JS_ORIGINAL_GA106.zip)

arutar, 2024-02-23:
@Usulyre Presumably someone tried to make rebar work, by default it does not work. There is a chance that the ports on the video card will work if they are 2xdisplayport 2xhdmi

Usulyre, 2024-02-23:
> @Usulyre Presumably someone tried to make rebar work, by default it does not work. There is a chance that the ports on the video card will work if they are 2xdisplayport 2xhdmi

Hi, the card has 3x displayports and 1x hdmi.

What is the bios you linked to?

Usulyre, 2024-02-23:
> @Usulyre This bios is designed for Jieshou RTX 3060m 2xDisplayport 2xHDMI. Perhaps the ports on your card will not work at all as on [Peladn](https://github.com/arutar/FrankenDriver/issues/7). But rebar works for you, it's very cool! So save your bios before flashing!

How about the 100% fan issue?

arutar, 2024-02-24:
@Usulyre It is unclear whether it is necessary to try

Usulyre, 2024-02-24:
> @Usulyre It is unclear whether it is necessary to try

Sorry but what does that mean?

Usulyre, 2024-02-25:
> @Usulyre In this firmware, the default fan speed is 30%, but perhaps there is no fan control on your card at all.

Ok, this seems to be true after some research.

So, I bought this at amazon: CRJ Micro PH 4-Pin PWM GPU Fan Header to Standard 4-Pin PC Fan Adapter Cable for Graphics Cards - 1.0 FT / 0.3 M

No i have gpu fan control. Although you have to set custom fan curve.

Still have no display ports working. Not sure what bios will work for this.

arutar, 2024-02-25:
@Usulyre It looks like you have the same problem as on peladn, no solution has been found at the moment

closed 2024-02-25 by arutar