arutar/FrankenDriver #46: Energy Efficiency questions? Idle Power Consumption & TDP information; Does driver used affect basic functionality?

Nuihc88, 2024-02-26:
I had previously been thinking about getting an Erying motherboard (with an integrated laptop CPU) and pairing it with a GTX 1650 (3-8W idle; 75W TDP) to build an either completely or semi- passively cooled desktop PC, but then just now ran into this and started wondering.....

1. Would using FrankenGPUs with laptop-chips always result in better energy efficiency?

2. Will using FrankenDriver impact Default Power Limits or returning from Sleep / Hibernation?

3. Does anyone have information on which Chinese laptop-to-desktop GPU has the lowest idle power consumption?

.....Knowing the TDPs of supported FrankenGPUs could be useful too, but since my primary use casee would be as a HTPC, it'll be idling most of the time.

arutar, 2024-02-26:
@Nuihc88 There is a [table of power consumption](https://www.techpowerup.com/review/nvidia-rtx-a2000/35.html) for desktop cards. Based on this table, the best option would be RTX A2000m. Based on this table, the best option would be the RTX A2000m. As for the processor, I advise you to consider the Ryzen 5 5500. Cheap and energy-efficient option.

Nuihc88, 2024-02-26:
> There is a [table of power consumption](https://www.techpowerup.com/review/nvidia-rtx-a2000/35.html) for desktop cards. Based on this table, the best option would be the RTX A2000m.

Indeed, RTX A2000 is certainly a far more powerful card than GTX 1650 and even has a lower TDP, but at least the desktop version of it seems to have ~2x the idle power consumption of some of the earlier GTX 1650s according to techpowerup. [1](https://www.techpowerup.com/review/palit-geforce-gtx-1650-kalmx/31.html) [2](https://www.techpowerup.com/review/palit-geforce-gtx-1650-stormx/30.html) [3](https://www.techpowerup.com/review/evga-gtx-1650-sc-ultra-black/30.html) [4](https://www.techpowerup.com/review/msi-geforce-gtx-1650-gaming-x/30.html) The problem with such charts is that i'm not sure how comparable desktop GPU power consumption metrics are to laptop GPU metrics. 

> As for the processor, I advise you to consider the Ryzen 5 5500. Cheap and energy-efficient option.

The Problem with that would be that Erying only makes their FrankenMotherboards with intel chips, so my best choices there are 13500H for maximum energy efficiency or 13900HK for maximum future proofing and performance. Either way intel's laptop CPUs are far less power hungry than their desktop CPUs.

arutar, 2024-02-27:
@Nuihc88 Mobile versions have certain limitations, for example, I have never seen a card whose fan rotation speed in idle mode would be 0; it is 30% by default. But when it is actively working, it consumes less electricity and heats up less.

arutar, 2024-02-29:
@Nuihc88 No

closed 2024-02-29 by arutar

Nuihc88, 2024-03-02:
> No

I guess this is in context of "_Does using FrankenDriver affect basic functionality?_"; If so then that is good to hear, but just for clarification what is the deal with '_Windows 528.24 (with support for changing the Power Limit)_' listed on the README.MD page?

arutar, 2024-03-02:
@Nuihc88 In version 528.24, nvidia has not yet disabled the ability to control power via the command line, in subsequent versions this feature is disabled for mobile versions of video cards