arutar/FrankenDriver #53: Linux drivers not installing / working

toastie000, 2024-05-28:
I tried for a few hours troubleshooting to try get the Linux drivers on the GitHub page to work, to no avail. They did not install under Mint, and they will not install under Ubuntu. They give the error 'Unable to build the NVIDIA kernel module'. I can give you the .log file if that's of any use. When I used step 2, they install, but the performance is awful, almost as if the drivers aren't working. Although, 'modinfo nvidia' shows the driver. Anything involving Vulkan just refused to work. 'vkcube' was running on the CPU with this driver (535.104.05).

I bought the boosty for the latest driver, which installed fine using 'step 1'. But, anything involving Vulkan refuses to open. Weirdly 'vkcube' runs on the GPU with the newer driver, though (550.78). Everything else uses OpenGL and runs awfully. I'm using a brand new Ubuntu installation (24.04 LTS), so I'm not sure if I've done something wrong or if it's the drivers.

arutar, 2024-05-29:
@toastie000 New driver version for Linux is now available for download from github. Have you installed the vlukan library?
`sudo apt install libvulkan1`

toastie000, 2024-05-29:
`libvulkan1 is already the newest version (1.3.275.0-1build1).`

I reinstalled the driver, and i got the error 'Unable to determine the path to install the libglvd EGL vendor library config files.' The driver will still install, though. I got this the first time I installed them, I just skipped over it though. I'm not sure if that's related, given the problem seems to be Vulkan. 

If I run steam in the terminal I get : 
`BInit - Unable to initialize Vulkan!`
`vkEnumeratePhysicalDevices failed, unable to init and enumerate GPUs with Vulkan. `

I'm not sure if It is the driver itself, or something that can be fixed, but I'll look into it tomorrow and report back.

toastie000, 2024-05-30:
Problem solved. It's just Ubuntu being shit as usual. `sudo apt install steam` refused to work on Ubuntu (worked on Mint, though) so I installed it through the 'App Center'. Somehow that caused Steam and Vulkan not to work together, I don't know why. After a few hours I reinstalled steam through the terminal with apt after adding the repository for it, all is working well now. Sorry to have bothered you.

closed 2024-05-30 by toastie000