arutar/FrankenDriver #57: Request support for 3500ADA, 4000ADA, 5000ADA

Sonicwanjie, 2024-06-25:
Request support for 3500ADA, 4000ADA, 5000ADA mobile card, is it possible to change the original driver?

arutar, 2024-06-25:
@Sonicwanjie Please send me device ids or gpu-z screenshots.

arutar, 2024-06-26:
@Sonicwanjie Try [this driver](https://disk.yandex.ru/d/aK5kzfavM2vR_g)  especially Nvidia Ada 3500

Sonicwanjie, 2024-06-28:
NVIDIA RTX 4000 Ada Generation Laptop GPU Device Id:10DE 27BA Subsystem Id: 103C 8B92

I modified the VB, with this driver can not install normally, with 10DE 0000 VB can install the driver normally. The graphics card is working. At present, there is a problem in the test, that is, the bandwidth of the graphics card has been locked at 4.0*1.1

arutar, 2024-06-28:
Can you provide the inf file?

arutar, 2024-06-28:
@Sonicwanjie In fact, nothing is blocked, it's just that the real performance can only be seen when the card is loaded.
[image]
[image]

Sonicwanjie, 2024-06-30:
[image]
The only problem at the moment is that the bandwidth is limited.

closed 2024-06-30 by arutar

Sonicwanjie, 2024-07-09:
[image]

arutar, 2024-07-09:
@Sonicwanjie Usually, these video cards have only one power consumption in the firmware. Those maximum minimum and default power are the same

Sonicwanjie, 2024-07-21:
2000ADA Device Id:10DE 28BB Subsystem Id: | 10DE 0000
3500ADA Device Id:10DE 27BB Subsystem Id: | 10DE 0000
4000ADA Device Id:10DE 27BA Subsystem Id: | 10DE 0000
5000ADA Device Id:10DE 2730 Subsystem Id: | 10DE 0000

Sonicwanjie, 2024-07-22:
NVIDIA_DEV.28BB.0000.10DE - NVIDIA RTX 2000 Ada Generation Laptop GPU，
Please help add this vend id to new driver 552.74.

arutar, 2024-07-26:
@Sonicwanjie [Driver](https://disk.yandex.ru/d/8FkXARHvAS7njw) When will you finally share one of these cards with me?

Sonicwanjie, 2024-07-29:
Feel free to share these cards with you and how to contact you offline

arutar, 2024-07-29:
You can write me via telegram @Frankendriver

Sonicwanjie, 2024-08-02:
@arutar My phone can't get messenger of telegram. I failed logon in telegram.

arutar, 2024-08-02:
@Sonicwanjie this is my email arutar@bk.ru

Sonicwanjie, 2024-08-22:
NVIDIA GeForce RTX 4090 Mobile，Device Id:10DE 2717 Subsystem Id: | 10DE 0000

@arutar pls add 4090m，thanks.

Sonicwanjie, 2024-08-27:
@arutar Expecting a new update, my operation on the payment page of the subscription channel failed. Are there any other channels subscribing to the new driver.

arutar, 2024-08-27:
@Sonicwanjie This device is already in the driver. Are you from China? Are you unable to buy a boosty subscription?

Sonicwanjie, 2024-08-28:
@arutar Is it possible to update the driver to the latest 560.94? In addition, I was unable to complete the payment in the subscription channel, changed 4-5 credit cards, and could not successfully complete the payment.

arutar, 2024-09-06:
@Sonicwanjie I have added a new payment system. [try this](https://app.lava.top/en/frankendriver)

Sonicwanjie, 2024-09-12:
NVIDIA RTX 2000 Ada Generation Laptop GPU，Device Id:10DE 28BB Subsystem Id: | 10DE 0000

@arutar I had buyed a boosty subscription. But 2000 Ada does not work.

arutar, 2024-09-15:
@Sonicwanjie i have fixed this issue re download last version of driver