arutar/FrankenDriver #58: Error code 10

ggeraleal, 2024-08-24:
I just got my RTX 3070m Ti from 51Risc, been using it for hours playing on emulators with the frankendriver 555.85.
But rn I tried opening Elden ring and suddendly the fans started ramping up a lot and I lost my video signal. 
Event Viewer show the following error code 10:
The GPU has been disconnected and this application may be unresponsive.
I restarted the pc and the signal came back, but is there something Im missing to prevent this from happening again? I did the clean install for EAC support, and installed the driver along side geforce experience.

arutar, 2024-08-26:
@ggeraleal Try the new driver version.

ggeraleal, 2024-08-26:
That same day I went to boosty and subscribed for the 560.94 drivers, and the same error appeared while playing Rage 2 with the new drivers. I opened rage 2, had the wrong language selected, closed it and changed the language and when I opened it up again, the same black screen with fans ramping up and the same error in event viewer appeared. The error hasnt happened since then, but it already happened once in both the 555.85 with elden ring and the 560.94 with Rage 2. I even played Elden ring at 2k 60fps for hours straight and couldnt get to replicate the error..

closed 2024-08-26 by ggeraleal