arutar/FrankenDriver #60: Can't Install On Fedora 40 KDE Linux - Can't get past kernel modules (550.78, "nvidia kernel module was not created")

NKkrisz, 2024-10-04:
Hello, I have a RTX 3080 Ti mobile I want to use but I can't get the drivers working.

Here are the steps I've done:

1. Uninstall old drivers with sudo dnf remove xorg-x11-drv-nvidia\*
2. Get driver from here and start with sudo after making it executable
3. Disable Nouveau and rebuild initramfs using the driver installer
4. Reboot
5. Start the driver installer again
6. Get stopped by the installer once kernel modules can't be installed

Attaching the log files.
[nvidia-installer.log](https://github.com/user-attachments/files/17262283/nvidia-installer.log)

My system:
[image]

binboupan, 2024-10-07:
Do you have the kernel headers installed?

NKkrisz, 2024-10-07:
Okay for some reason in the past if I remember correctly the propretiary drivers didn't work but now I tried installing it again from Nvidia's website and it seems to be running fine so far? Weird...

https://www.nvidia.com/en-us/drivers/details/232672/

[image]
[image]
[image]

arutar, 2024-10-07:
@NKkrisz It looks like a newer version was installed after the patched frankendriver installer was run.

NKkrisz, 2024-10-07:
> @NKkrisz It looks like a newer version was installed after the patched frankendriver installer was run.

Yeah I couldn't get the Frankendriver working so I tried installing the propreitary one again which seems to be working so far.

Anything different between the Frankendriver that isn't in the propreitary one?
What things would you recommend testing / looking out for to make sure everything is working fine?

Thanks.

krisives, 2024-10-07:
This could have been related to FrankenDriver not installing: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/642

arutar, 2024-10-07:
@krisives @NKkrisz @binboupan I have posted the latest drivers for Linux

closed 2024-10-07 by arutar