arutar/FrankenDriver #61: Error code 43

namrekayt, 2024-10-14:
[image]

arutar, 2024-10-14:
@namrekayt This message may appear due to a faulty video card. Please send screenshots of GPU-z

namrekayt, 2024-10-15:
[image]
[image]
[image]

arutar, 2024-10-15:
@namrekayt try to delete all video drivers with DDU and then reinstall the driver

namrekayt, 2024-10-15:
Still, the same thing happens

arutar, 2024-10-16:
@namrekayt If it is possible to connect to another computer - try. But if it does not work on another computer, then the problem is in the video card itself.

closed 2024-10-19 by arutar

arutar, 2024-11-03:
@namrekayt 
https://youtu.be/qHzO_mRqtAM?si=jrdwlWNSfGehIyxG
Try to install manually 1660 super
And there is a topic on techpowerup
https://www.techpowerup.com/forums/threads/solution-for-black-screen-rtx-3060m-china-gpu-ninja-jieshuo-and-more.315137/