arutar/FrankenDriver #63: Request non DCH Driver for Quadro Franken Card on Server 2016

dateno1, 2024-10-31:
Original Quadro A2000 work well on Server 2016

but FrankenDriver for 2016 is not exist (FrankenDriver is all DCH Type and It can't installed on 2016)

Can you add Driver for old OS or help to create new driver for 2016 by myself?

arutar, 2024-11-01:
@dateno1 do you need 538.95 driver?

dateno1, 2024-11-01:
> @dateno1 do you need 538.95 driver?

Yes I need quadro 538.95 for Franken A2000M

arutar, 2024-11-02:
@dateno1 
Try [this driver](https://disk.yandex.ru/d/_Z-4lZS95NKn5g). And if it works fine dont forget to say thanks. I spent a lot of time on the patch

dateno1, 2024-11-03:
> @dateno1 Try [this driver](https://disk.yandex.ru/d/_Z-4lZS95NKn5g). And if it works fine dont forget to say thanks. I spent a lot of time on the patch

Test on BD790i
CPU+ Board : Minisforum BD790i
RAM : 32G*2
Power : 700W
VGA : A2000M
SSD

It hasn't some of AMD Drivers for 2016 but work well (Cheap&Fast board)

Test with Server 2016/10 22H2 (Lastest)

both Updated to Lastest version

Server 2016 hasn't any UWP components (I was add It with script) -> Manually Install NVIDIA Control Panel by downloaded appx)

Test result : All work well on both version

Nice :D

Thanks for help&fast response

Are you has 'buy me a coffee' or 'paypal' account? (I don't want to do subscription but want to send some of money)

closed 2024-11-03 by dateno1

arutar, 2024-11-03:
@dateno1 https://app.lava.top/ru/frankendriver?donate=open

dateno1, 2024-11-03:
> @dateno1 https://app.lava.top/ru/frankendriver?donate=open

OK i sent $20

Thanks for help

Now I can use RTX Video Super Resolution on old OS :D