arutar/FrankenDriver #65: Mllse RTX3060m 12Gb

ByWallkroft, 2024-11-02:
Hi. I got some problem. Driver  is partially crashing after starting benchmarking/games (look photo)/ Ind i got black screen in games ofc
1. 
[image]
2. 
[image]
3. 
[image]
Sorry for photo (none screenshot), i got troubles with making screenshot at the test stand

arutar, 2024-11-03:
@ByWallkroft сложно сказать что за проблема. Я бы пробовал на другом компе (рабочем) если проблема повторяется надо отдавать в ремонт видяху. 
А так может быть все что угодно
БП
Винчестер 
Память

ByWallkroft, 2024-11-04:
> @ByWallkroft сложно сказать что за проблема. Я бы пробовал на другом компе (рабочем) если проблема повторяется надо отдавать в ремонт видяху. А так может быть все что угодно БП Винчестер Память

судя по всему один из банков памяти неисправен

ByWallkroft, 2024-11-04:
Делал на заведомо исправном компе под 11 виндой, где у меня CMPшка стояла. 
До этого на стенде с 10 виндой та же шляпа. 
Даже упоролся, нашёл того единственного несчастного (пока ждал ответ), что такую же карту продавал, спросил, на каких драйверах у него работало - всё едино.
Так что тему можно закрыть, а карта поедет обратно к продавцу

closed 2024-11-04 by arutar