arutar/FrankenDriver #66: Windows 7 Version

q228777, 2024-11-09:
Could you make such a driver for Windows 7, even if it's not the latest version? Or can you provide instructions on how to modify the driver in Frankenstein?

arutar, 2024-11-09:
I'll see what I can do about it. Why do you need a version that is no longer supported?

q228777, 2024-11-09:
I don't use Windows 10. Any version of the driver for the 3070m would work for me.

arutar, 2024-11-11:
@q228777 Try [this version](https://disk.yandex.ru/d/BZqF6wjKNopsOg)

q228777, 2024-11-11:
Thank you very much, the driver is working. However, to bypass the error "the signing certificates are not installed", I had to enable the mode for disabling driver signature enforcement and install the driver through the Device Manager. Unfortunately, the NVIDIA Control Panel did not appear. Do you know how this can be resolved?

closed 2024-11-11 by arutar