arutar/FrankenDriver #68: Very low FPS, High Core Usage

darkanubis0100, 2025-01-09:
I am having a problem with my RTX 3080 Ti from "Aliiphone".

It turns out that it has strange behavior when using the drivers, if I do LLM inference on things like LM Studio, the performance may well be 52 tk/s or 3 tk/s. It's quite strange since with such a core and 512 GB/s of bandwidth it should be around 55-60 tk/s and at no time should it drop.

But then I go to the Gaming section and it is even worse, the core is set to 100% (2.1 GHz frequencies) and the FPS at no time exceed 20 FPS. Things like Dead Space Remake are completely unplayable (7-12 FPS).

I was curious about a question... Does that Frankenstein driver conflict with other RTX? I have an RTX 3070 Ti (From MSI) that is also using the Frankenstein drivers (For some reason I can't get the normal driver to install on the normal GPU without the AliExpress one being affected).

arutar, 2025-01-10:
What version of driver are you using? Try [this](https://disk.yandex.ru/d/xX-n0epD_CDnVg) version.
There should be no conflicts. Have you changed the default frequencies of the video cards?

darkanubis0100, 2025-01-10:
> @darkanubis0100 What version of driver are you using? Try [this](https://disk.yandex.ru/d/xX-n0epD_CDnVg) version.
> There should be no conflicts. Have you changed the default frequencies of the video cards?

Hello!

After doing several measurements in GPU-Z and performed tests comparing the mentioned there (In GPU-Z) with periods of inactivity and cold load, I have realized that this is most likely a hardware problem, specifically in the VRM.

Today I talked to the person who sold me the board and he gave me authorization to open it (under warranty) and examine what problem I can see. So for now everything points to be a hardware failure and not a software failure.

closed 2025-01-10 by arutar