arutar/FrankenDriver #69: Is there going to be 572.16 version of FrankenDriver?

Stradivario, 2025-02-05:
Hey guys!
Thank you very much for the project!
I really appreciate your work!

Just curious is there any plans for porting the new 572.16 driver with DLSS 4 support for Franken Cards ?

I recently purchased 3080Ti M 16GB and i am happy person i got it for 300 Euro which is a bargin for this GPU!

I just want to have some more frames out of the GPU using DLSS 4 since i understand there has being a various optimizations.

Thanks in advance to everybody involved in this project!

@arutar

arutar, 2025-02-05:
@Stradivario Hello, the latest drivers are available by subscription. This version will be released for free later.

Stradivario, 2025-02-05:
@arutar 

Amazing!

I have subscribed to your Feed lets see what you have there :))

closed 2025-02-10 by arutar