arutar/FrankenDriver #73: Linux 570.144 or 570.153.02 Release?

NKkrisz, 2025-05-19:
Hi, just wondering if you could please update the Linux drivers to be a bit more up to date?

I'm having a hard time installing 550.120 on Fedora 42 KDE Edition, it has errors while building the kernel modules - to me it looks like it wants to use things from old kernel perhaps. The kernel I have installed is: Linux 6.14.6-300.fc42.x86_64

Some kind of tutorial on how to edit the drivers yourself would be great, but I understand if you wouldn't want to do it.
I tried making changes myself but couldn't find great resources. I'm using an RTX 3080 Ti Mobile (PCI ID - 10de:2240 + Subsystem ID - 0000).

Any help would be greatly appreciated, thank you in advance and hope you're having a good day :)

arutar, 2025-05-20:
done.

closed 2025-05-20 by arutar

NKkrisz, 2025-05-20:
Thank you for the quick response, I sadly still can't get the GPU to work correctly.

I've tried installing the drivers on my own and using a tutorial but both don't recognize the GPU (`nvidia-smi`) after doing so.
The driver itself installs successfully, I have no idea what's causing the issue.

The tutorial I've followed is the following - [text version](https://www.if-not-true-then-false.com/2015/fedora-nvidia-guide/) / [video version](https://www.youtube.com/watch?v=e90qwj6wsC0)

I've got Secure Boot disabled and would like to keep using Fedora 42 with KDE Plasma and Wayland if possible (X would be fine as well if Wayland can't work).

Do you have any ideas that could help?
What logs should I attach that could be helpful for debugging this?

Thanks again.