arutar/FrankenDriver #77: Code 43 on Windows

JackieTheBunny, 2025-06-27:
I'm getting a Code 43 for the driver, trying to reinstall it now

JackieTheBunny, 2025-06-27:
still doesn't work upon reinstallation

arutar, 2025-06-28:
Please send a gpu-z screenshot

JackieTheBunny, 2025-06-28:
I can't find mine on that list, but mine is an RTX 3060 16 GB found in a Tank 03 Acemagic mini PC

arutar, 2025-06-29:
@SewageServices @JackieTheBunny It is also possible that the card and platform are not compatible, and a different manufacturer's fransce-shtein will work. From the screenshot, it is clear that the video card's BIOS does not support UEFI

SewageServices, 2025-06-29:
Weird, because it used to work up until recently.

JackieTheBunny, 2025-06-29:
> Weird, because it used to work up until recently.

Same, my driver was working perfectly, then suddenly code 43'd

> You need to enable above 4g in Bios. And try to remove all previous drivers using DDU

The problem is, removing the drivers from my PC with DDU breaks things. I can't install it back, it fails to install

JackieTheBunny, 2025-06-30:
[image]

arutar, 2025-07-04:
@SewageServices @JackieTheBunny This is a hardware error, not a driver error, and the 3060 stopped working on your platform. I think if you try running the cards on another computer, everything will work fine

JackieTheBunny, 2025-07-04:
> [@SewageServices](https://github.com/SewageServices) [@JackieTheBunny](https://github.com/JackieTheBunny) This is a hardware error, not a driver error, and the 3060 stopped working on your platform. I think if you try running the cards on another computer, everything will work fine

Soooo... Essentially it boils down to my $600 computer being a piece of shit?

arutar, 2025-07-05:
@SewageServices @JackieTheBunny I can't say for sure. But I can't offer a specific solution either. In my opinion, something has happened to the bios settings. Try resetting the bios settings or finding the necessary ones. Sometimes, during an update, windows may download and install a new firmware for laptops. I don't know if this could have happened to you.

arutar, 2025-07-22:
I am closing the issue as there is a problem with the host hardware.

closed 2025-07-22 by arutar